electricityforprogress/BiodataFeather #1: Update on ESP32-S3: compiled!

ooktown, 2022-07-02:
Hi Sam - I wanted to give you an update on my progress with the BiodataFeather code.

**The good news:** I've compiled the code for the ESP32-S3 chip, and uploaded it successfully!
**More good news:** When I first plugged the ESP32-S3 into my Mac, it was immediately recognized as a "keyboard" (native USB works!).
**Even more good news:** I can connect to the board via BLE.

**The bad news:** It's not connecting to the network. Here's the error it's giving:

`[WiFiUdp.cpp:219] parsePacket(): could not receive data: 9`

I'll be digging into it to see if I can get it to connect to the WiFi successfully. I'm also going to try adding the WiFi Manager code so I can let other people use the interface on their network.

electricityforprogress, 2022-07-02:
Oh yeah, wifimanager is a great option.  I've been 'rolling my own' AP with wifi credentials, but don't have anything running well enough to share as of yet.

ooktown, 2022-07-02:
FYI, here are the stats from compiling the code for the [ProS3](https://esp32s3.com/pros3.html) (ESP32-S3) from Unexpected Maker:

```
Retrieving maximum program size .pio/build/um_pros3/firmware.elf
Checking size .pio/build/um_pros3/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.7% (used 67828 bytes from 327680 bytes)
Flash: [==        ]  18.7% (used 1224517 bytes from 6553600 bytes)
Building .pio/build/um_pros3/firmware.bin
esptool.py v3.3
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
```
I'm not sure how that will translate for the Adafruit ESP32-S3 Feather which has only 4MB Flash and 2MB PSRAM.

I also compiled a WebServer Portal as a Proof-of-Concept created by [Khoi Hoang](https://github.com/khoih-prog/ESPAsync_WiFiManager), and it worked with no problems. Next challenge is to integrate the code into the BiodataFeather code base. And then cross my fingers it fits into the Adafruit ESP32-S3 Feather flash! :-D

I'd be glad to share my code once I get it up and running.

closed 2022-07-04 by ooktown

electricityforprogress, 2022-07-04:
This is all really great thank you so much for putting in the legwork to get the wifi server running!  I have a couple other newer versions of the code with minor variations and features.  If you continue to modify the V005 code then I can merge stuff together.  Please share your codebase

I had setup the 'samFader' objects to use the ESP32 channels, but it is more of a hack.

As you have seen, my Menu routine is absolutely horrible, this is something which I will also have refinements for.

There are a couple of places where the serial, bluetooth, and wifi MIDI notes and CC controls are set and updated, it should be easy to include a block of code for the USB MIDI and pop in a MIDI flush.

I'll spin up a biodata board with the Adafruit ESP32-S3 and look at the fading leds.

ooktown, 2022-07-04:
Here ya go! I'm not a coder by trade. Just a hacker... I hope I put enough comments in there to guide you through my changes.

Theoretically, this code should compile and run on the ESP32 feather you have, so you can try it out right away:

Since you're make tweaks to the code, I'll wait to pull down an updated version from Github if you can get the WiFi manager running on the ESP32 Feather.

Let me know if you want my help!

Take care,

Stuart Yoshida

Ukulele Instructor and Performer
"A Seriously Fun Place of Learning"
www.ooktown.com <http://www.ooktown.com/>
***@***.*** ***@***.***>
Podcast: www.ooktown.com/podcasts

ooktown, 2022-07-04:
Made some corrections to add back in your LED code:

Stuart Yoshida

Ukulele Instructor and Performer
"A Seriously Fun Place of Learning"
www.ooktown.com <http://www.ooktown.com/>
***@***.*** ***@***.***>
Podcast: www.ooktown.com/podcasts