electricityforprogress/BiodataFeather #2: BiodataFeather board question: substitute for 3.9K timer resistor? Also, NE555 vs LM555 chip tradeoffs?

ooktown, 2022-07-02:
Hi Sam - I had a question on a resistor value for the LM555 timer chip: what can I substitute for the 3.9K resistor? I have 3.3K or 4.7K resistors, but unfortunately I don't have any 3.9K resistors. Can I substitute one of those values for the 3.9K?

Also, I'm ordering some LMC555CN timer chips, but I have lots of NE555 chips. I was planning on temporarily using the NE555 timer chip in the socket while I wait for the LMC555CN chips.

What are the drawbacks for using the NE555 chip? I read that the NE555 chip is less stable over temperature and time. Will this cause glitching in the biodata?

Thanks again for all the great work on the BiodataFeather! It's awesome to have this as an open-source project.

electricityforprogress, 2022-07-02:
The Biodata device uses a 555 timer to detect electrical conductivity between two probes.  The Resistor and Capacitor network set the 'rate' at which the 555 timer detects changes.  You can substitute a similar value resistor without huge impact to the detection. 

When it comes to 555 timers not all are built the same!  The important thing with this circuit is that we use a 'low power' timer which will operate at the 3.3v of the ESP32.  This leaves VERY FEW options, and the LMC555 is the primary timer which does all of the magic.  Now I already know there are no through-hole LMC555 timers available anywhere in the world, and many chips which are out there are 'fake', the chip crisis is looming.

I always say "experimentation is proof", you can pop a timer in and try it out, but sadly the NE555 needs 5v by the spec sheet.  From a circuitry standpoint, you very well might be able to get a 5v 555 timer running through an optoisolator and send that signal into the ESP32.  There are so soo soooo many 555 timer chips, and many low voltage models.... I have only found that the LMC555 does the magic best, but you are welcome to try other timers and share your results!

One other part of the circuit is a DC-DC voltage isolator, this is very important if you are connecting to USB.  We want to 'galvanically isolate' the 555 timer from any ground or certainly from your Laptop power supply.  Also if your audio output on the laptop is connected to a stereo system ... this creates a Ground Loop.  Be aware that this will happen, your circuit may work just fine but if you can properly isolate the detection circuit then you will get maximum sensitivity.  All of this is why I use Bluetooth and rechargeable batteries!

ooktown, 2022-07-02:
Thanks for all the feedback, Sam!

I'll substitute a 4.7k resistor for the 3.9k.

I have some supposed LM555 timer on order, but they may very well may be fakes. We'll see how it operates when I stick one in the circuit.

I will definitely keep the isolator in the design to make the input as clean as possible.

And I also prefer using BLE with batteries!

ooktown, 2022-07-04:
Once I get some parts together, I'll will try out some 555 chips in the socket to see what works. For now, I'll close this issue.

closed 2022-07-04 by ooktown