electricityforprogress/BiodataFeather #3: Which pins are beings use for Serial1?

ooktown, 2022-07-04:


electricityforprogress, 2022-07-04:
It is mapped to IO17 for tx on the adafruit feather

ooktown, 2022-07-04:
Thanks! I'll give that a try.

electricityforprogress, 2022-07-04:
You could also just hack out all the hardware serial code if it's causing a compile issue for the s3. Unless you intend to use hardware serial rather USB.  I'll politely apologize for my extremely messy menu coding. But hey that's just how it rolls... I'll refactor this later ... as the saying goes

ooktown, 2022-07-04:
Ok, found the issue: the LED fader routine in the setup() is causing the crash. I took it out, and the code is running!

So we can close this issue.

WiFi captive portal: working (tested, can connect to WiFi from portal)
BLE: working (untested, but showing up on Bluetooth sniffer and I can connect)
Serial1: working (untested, but it's not crashing :-) )

While I wait for the parts for the galvanic interface to arrive, I'll work on debugging the LED fader function and try to get MIDI over USB working.

And no worries about the nested menu code. I just "prettified" it by formatting it enough so I could grok what was going on. :-D

closed 2022-07-04 by ooktown