electricityforprogress/BiodataFeather #4: Optoisolator alternative?

inktrap, 2022-07-30:
This Optoisolator:

```
CNC1S171S0LF	Panasonic Electronic Components	OPTOISOLATOR 5KV TRANSISTOR 4DIP
```

seems to be obsolete and not manufactured anymore:
https://www.digikey.de/en/products/detail/panasonic-electronic-components/CNC1S101/970631
However I found the **R0** instead of the **S0**:
https://www.digchip.com/datasheets/parts/datasheet/2099/CNC1S171R0LF.php

The specks seem to be the same: https://www.findchips.com/parametric/Optoelectronics/Optocoupler?term=CNC1S171S0LF&Manufacturer%20Part%20Number=cnc1s171*

inktrap, 2022-07-30:
Also the pc 817 is all over the place … and seems fitting … https://www.findchips.com/parametric/Optoelectronics/Optocoupler?term=pc817&Manufacturer%20Part%20Number=pc817

ooktown, 2022-07-30:
We have successfully used the [LITEON LTV-816](https://www.digikey.com/en/products/detail/liteon/LTV-816/385830?s=N4IgTCBcDaIIwDYAMBaOBmZKCsKByAIiALoC%2BQA) on a batch of four Biodata boards as an alternative part. YMMV.

The boards are for a music project I'm collaborating on, and all four boards using the LITEON optoisolator are operational.

electricityforprogress, 2022-07-31:
Excellent! I just got home from a beach weekend and I'm super glad to hear
that you have everything up and running. Yeah most of the isolators with 3v
operating voltage will work with this module. Most of the parts are pretty
generic really only thing that is important is the 555 timer it seems. I
have tried using many alternatives and other low power timers and the LTC
555 from Texas instruments is alway the best choice for me.

inktrap, 2022-08-03:
Hey :) thank you for your answers! It took some doing but i finally ordered all the components, and I double checked that I have the right ones. I gues 20AWG wire would be sufficient? Also I just hope that the pcb will work, but that is an experiment. First time ordering a PCB :D I guess I will build this on a breadboard first to get a feeling for it, then on one of the PCBs.

electricityforprogress, 2022-08-03:
Yes really any wire will work for the electrode leads, it is easy enough to
recycle some headphone cables or use some flexible two conductor.  The
commercial wires on headphones and electrodes are probably way thinner than
20 awg.

What are your concerns about the PCB?  If you used my gerber files and sent
to a board manufacturer then yeah everything should work easily.  I use
NextPCB, but most people see the PCBway adverts or JLCpcb ... if this is
your 'first time customer' at a PCB house and you only are buying 5 of one
design, the boards should be 'free' and just pay shipping ...

Yes yes yes, definitely build on breadboard first to make sure you have all
the right parts and values ;) also make sure to use a socket for your 555
timer ... if you have a problem in assembly at least the 555 timer is easy
to salvage!  But if you watch my soldering and assembly videos the process
should be simple for you.

Sam Cusumano
Electricity for Progress <https://squareup.com/store/electricityforprogress>
***@***.***
267-230-1802

inktrap, 2022-08-03:
> Yes really any wire will work for the electrode leads, it is easy enough to
> recycle some headphone cables or use some flexible two conductor.  The
> commercial wires on headphones and electrodes are probably way thinner than
> 20 awg.

Ah, I bought some electrode leads and some 20AWG wire for another
project but won't be needing it because everything is directly soldered
to the board. Next time I am going to recycle something and build my
own. I think I should be able to do this with the video tutorial … that
was very well explained, so thanks for that! :)

> What are your concerns about the PCB?  If you used my gerber files and sent
> to a board manufacturer then yeah everything should work easily.  I use
> NextPCB, but most people see the PCBway adverts or [...]

Oh I got some 2-3€ discount and paid like 2€ for the pcbs and 5€ for
shipping and I chose black as the pcb color which increased the price a
bit. Also they have to pay import taxes that they will price in.

I didn't use your gerber files, just the once I exported from eagle.
Since I have never done this before, it was exciting, and that is where
my concerns come from … there is no explicit reason that I know of why
this could go wrong. Maybe it's like flying for the first time?

> Yes yes yes, definitely build on breadboard first to make sure you have all
> the right parts and values ;) also make sure to use a socket for your 555
> timer ... if you have a problem in assembly at [...]

Yes, I read the tips and also the forum and ordered some sockets.
Since this is my first project I have to learn how/where I can salvage
things … as of now I am just happy that I bought the right parts ;P

Thanks, again! :)

ooktown, 2022-08-03:
I will second the advice of having the 555 timer in a socket. It allowed us to try different chips and will make replacement easy, if needed.

Also, before having the parts soldered to the PCB, I randomly ohm'd out about five traces to give me confidence the board was manufactured correctly.

Good luck! It's a fun project.

inktrap, 2022-08-03:
Which timers did you try, different versions of the 555 or completly different timers, and what were the effects? I recently found out that there are also adjustable timers, that give some more flexibility, something like this: https://duckduckgo.com/?q=arduino+adjustable+555+chip&t=ffab&iar=images&iax=images&ia=images

And yes, I chose this project because it is fun and also beginner friendly, very well documented and it also helps that I love music, plants, and the creative and educational aspects, so … awesomeness all around ;P Another awesome project that I wanted to mention is the pedalpiv2 that could be used during a live-performance that could accompany the daw with biosonic music … https://github.com/thawney/PedalPiV2

electricityforprogress, 2022-08-04:
When the 'chip shortage' was first talked about, I tried a couple different alternative low voltage timer on a simple breadboard circuit with an ESP32 Feather.  The results were surprising ...
1. TS555CN https://www.amazon.com/gp/product/B07FJ6Y3VR/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
2. ICM7555IPAZ https://www.digikey.com/en/products/detail/renesas-electronics-america-inc/ICM7555IPAZ/821488
Both of the timers tested 'work perfectly fine' as microcurrent sensors when attached to plants, BUT when the electrodes are not connected to anything the entire ESP32 would freeze up!  Totally strange and i poorly studied the situation.  Why does this matter, well something strange is happening and i can't afford the system freaking out in the idle state.  I ended up just bailing on alternatives and 'only' using LMC555's

I am fairly certain both of these chips would function in a full circuit, with proper DC power isolation and an Opto on the output, more experiments would prove the outcome.  Secretly I want to get Texas Instruments sponsorship ;)

inktrap, 2022-08-04:
Mmmh that seems strange and you mentioned this in the forum already.
Out of curiosity … What would an adjustable timer (not) do?

> Secretly I want to get Texas Instruments sponsorship ;)

Haha :) that would be awesome! :)

But seriously, I hope you don't mind me going down the rabbit hole of YT
channels/content in a github issue and sharing my perspective …

If your channel has a certain size I think you could easily get the
usual sponsors, like NordVPN, Curiosity Stream, aso but I never saw TI
or a similar company sponsor someone, I guess it wouldn't affect their
bottomline too much and they sell chips anyways?!

I would love to see another "I have fun with music and electronics"
channel with proper sponsorship and regular unusual and creative
(arduino) projects that are explained well.

F.e. It was awesome when I found this one:
https://www.youtube.com/c/M%C3%BAsicoNerd which also seems like a good
way to promote his courses on midi-controllers and how to make musical
devices with arduinos. Fair enough.

If you have a look at e.g. his theremin (that I want to build after the
biosonic device) video (https://www.youtube.com/watch?v=CWPtxGrBlU8),
there are many questions that pop up in the comments (which screws did
you use, what are the dimensions of the case, aso aso) and, while his
production value is pretty high and the video editing is good aso, it is
a video that introduces the device than to follow along with a build.
Your project and your explanation was way easier to follow.

In contrast I also liked this video a lot which is more a piece of art …
but that connected with me and I really like his style and I watched it
way too many times ;D
https://www.youtube.com/watch?v=gMH9M2vcFZk

electricityforprogress, 2022-08-04:
For an Adjustable timer I really have no idea, i mean the 555 is an adjustable timer and the LMC is just the cmos low power version.  I think that the device is using a minor 'trick' pushing pulse widths based on microcurrent changes across the threshold and discharge pins.

I had entertained the idea of doing some more 'monetized' content on youtube, but then i realized that other organizations can simply claim the monetization due to strange audio algorithms.... and so honestly fu** that right in the goat a** , I'd rather not do the whole 'ask youtube and sponsors to pay for the content based on ad revenue and copyright hunters.  what grotesque economy!   (grumble....as i pay my $10 monthly youtube-red account to avoid ads)

great comments, cheers!

inktrap, 2022-08-04:
> For an Adjustable timer I really have no idea, i mean the 555 is an adjustable timer and the LMC is just the cmos low power version.  I think that the device is using a minor 'trick' pushing pulse widths based on microcurrent changes across the threshold and discharge pins.
> 

Ah I had no idea what the timer does … ofc. it sends pulses for measuring, I read up on that, but since I am brand new to electronics I assumed it was somehow fixed. I still have to learn a lot … and I guess I will turn to youtube for that …

> I had entertained the idea of doing some more 'monetized' content on youtube, but then i realized that other organizations can simply claim the monetization due to strange audio algorithms.... and [...]

I mean the question is if you like the idea to be someone who does more monetized or streamlined content. I guess people who do that like doing it despite youtube's "stance" on content. With more professionalization (as in profession) stuff becomes somewhat interchangeable too, and the artsy-video I linked is much more authentic/fascinating, at least to me. I don't want to put content creators down when I write that, those two types of videos are simply so different that the videos are two different things.

> (grumble....as i pay my $10 monthly youtube-red account to avoid ads)
> 

May I interest you in https://ublockorigin.com/ and https://github.com/ytdl-org/youtube-dl then?
And if you can install f-droid on android https://github.com/TeamNewPipe/NewPipe/ is awesome-sauce :)
A good network-based solution is https://pi-hole.net/

> great comments, cheers!

thanks :)

ooktown, 2022-08-05:
> Which timers did you try, different versions of the 555 or completly different timers, and what were the effects? I recently found out that there are also adjustable timers, that give some more [...]

We used some LMC555CN knock-off parts. It was a gamble that paid off. YMMV. Authentic TI parts, as Sam has mentioned, are hard to come by. This chip shortage really sucks!

inktrap, 2022-08-05:
> We used some LMC555CN knock-off parts. It was a gamble that paid off. YMMV.

I ordered some LMC555CN timers via aliexpress and can provide feedback wether that worked for me too. I am optimistic ;P. Side note: I asked about ROHS compliance and they had no idea about that because "they are just as distributor". Do I have to worrry about ROHS? At least I made sure that the PCB is ROHS+lead-free.

Also about the chip-shortage: I mentioned pedal-pi earlier which requires an raspberry pi zero … and you can only buy them used and for around 6 times the original value. Maybe, if it is still a hobby, the solution is to do less and to be happy with less? I will wait with my pedal-pi build :).

closed 2022-10-01 by electricityforprogress

ooktown, 2022-10-11:
FYI: I used the ECAD module in Fusion 360 to successfully open the design files and export Gerber files for manufacturing at a vendor.

And I also love having colorful boards! We have so far printed purple and white boards. Both colors look great!

inktrap, 2022-10-11:
When I'll finish it (and it hopefully works) I'll send you a picture of
the black one :D