electricityforprogress/BiodataFeather #5: Request: code to enable double-click, long-click on buttons

ooktown, 2022-08-05:
I'm looking to add the ability to enable and disable some additional options for the Biodata Feather interface (extra CC messages, AP setup), and it would be great to turn on/off these options with a double-click or long-click.

It appears that the samButton class was partially set up to support these functions, but I don't think it was fully enabled. Would you post the additions/updated samButton class to add these abilities?

electricityforprogress, 2022-08-05:
Hmm, perhaps ;) but unlikely in any relative timeframe.  I only spun my own for portability.  Thinking about it, this should be trivial to implement.  If i can think of something (or you can suggest something) for which long press would be valuable then i might do it while watching Critical Role tonight.  

Now i know that my menu code is totally the pits.  Describe how you want the AP to work, cause i have a couple ideas too.  I think it could be done in the wifi sub menu as a third option. ...  but i digress.

Give me one good reason to add long press and I'll do it ;)  - Cheers!

ooktown, 2022-08-05:
How about TWO reasons :-)

1. Turn on/off CC 84 for extra control over MIDI sounds. Once we get it going, we'll share what we're doing. For now we want to keep it under wraps ;-). Suffice to say, it should be very cool when we get it working with CC 80. This could be an option when in MIDI channel mode. FYI: I already have CC 84 working along side CC 80. I'm using a third-degree polynomial equation to give CC 84 more "wiggle" relative to CC 80.

1. Turn on/off WiFi STA mode. I am currently using the WiFi option to enable the AP mode to configure parameters, including the MIDI channel. (We will be adding more parameters as needed.) The AP eventually goes away for security purposes. I would like to be able to enable/disable WiFi connectivity to a network (STA mode) on-demand from the "Green" mode for future use with MQTT or to control LEDs by sending data to a [Pixelblaze](https://www.bhencke.com/pixelblaze), for example.

And double-click would work just as well as long-click, as far as I'm concerned.

ooktown, 2022-08-05:
And one more reason...

3. Set the sensitivity to be static at whatever setting the pot is at, and then use the pot to control a MIDI parameter.

closed 2022-10-01 by electricityforprogress