electricityforprogress/BiodataFeather #6: I ordered an ESP 8266 by mistake, do I need the Mhz?

inktrap, 2022-08-21:
I read that [the main difference is Mhz](https://www.utmel.com/components/esp32-vs-esp8266-which-is-better?id=921): 

> The ESP32 is a dual-core 160MHz to 240MHz CPU, whereas the ESP8266 is a single-core processor that runs at 80MHz.The ESP32 and ESP8266 are both fantastic Wi-Fi modules.

Should I just by an [ESP32 from china](https://www.aliexpress.com/item/1005001757645011.html?mp=1) and send the [Feather ESP8266](https://www.reichelt.de/adafruit-feather-huzzah-ada-fb-huzzah-p235448.html) back?

electricityforprogress, 2022-08-21:
Sadly no the 8266 is an earlier version and has many fewer capabilities. We
use Bluetooth on the ESP 32. The 8266 can certainly run the code and could
output cereal data, and could probably even handle Wi-Fi RTP MIDI. But I do
not have a code base tested for that module and I'm sure there will be many
tweaks required.

I recommend trying to get an ESP 32s3 development board which does
everything the older esp32 does plus has USB midi. I have a new code base
for the ESP 32s3 on my GitHub.

inktrap, 2022-08-21:
Ah alright, I am trying to return it then! What does the s3 stand for? [Would one of these work?](https://www.aliexpress.com/item/1005001757645011.html?mp=1) … They are also much cheaper …

electricityforprogress, 2022-08-21:
At $0.96 I would be very skeptical of that specific board. I really don't
know how best to advise you here, any esp32 dev board should work, you just
have to hook it all up right.

The S3 is a new model of the ESP 32.

You pretty much have to figure it out from this point though I'm not going
to map the pins and answer a whole bunch of questions about weird boards if
you know what I mean, no offense my friend.

inktrap, 2022-08-22:
Hey :)

my apologies if that was an annoying question. I thought it would have an easy yes/no answer but it is much more complicated, which I understand now.

However, this level of complication leads to  … a let's say steep learning curve. Many abbreviations, a lot of subtle differences, boards have variants that behave differently … so it is easy to make mistakes. This is my first project and I have all the other parts. It would be quite sad if I bought the wrong one again, I am sure others also appreciate it that I am asking for one more detail regarding the board: does flash/sram matter? The ones I found are from 14-18€, so they have a reasonable price:

 - ESP32-S3-WROOM-1-N8R2(8M Flash/2SRAM)
 - ESP32-S3-WROOM-1-N8R8(8M Flash/8SRAM)
 - ESP32-S3-WROOM-1-N16R8(16M Flash/8SRAM)

The distinction between [ESP32-S3-WROOM-1, ESP32-S3-WROOM-1U, ESP32-S3-WROOM-2](https://www.espressif.com/en/products/modules/esp-wroom-32#ESP32-S3) (+ mini variants) doesn't seem to matter and I would go with the `1`.

It would be really helpful, and good from an didactical perspective, if that could be included in the BOM or documentation?! If not … feel free to close this issue, then I would just buy the largest one. I am also fine with choosing another project to learn the basics, so no questions about pinouts.

Last but not least: I could get my money back for the 8266 and they don't even want the board back?! Wow!

Thank you for your time and attention :)

electricityforprogress, 2022-08-22:
Sorry i get a bit twitchy when it comes to using different Dev boards.  I
teach electronics and I encounter questions about using other boards all
the time from students, and each face the same challenges.  Migrating
between Dev boards is fairly easy for an experienced programmer, and can be
very confusing for a beginner.

You raise an important point here.  There are many variations of the ESP32
board, and each variation will require a specific board definition file in
Arduino and each will have different parameters for compilation.   The
simple BOM shows the Adafruit ESP32 Huzzah Feather (and the latest version
of the code uses Adafruit ESP32-S3-2MB https://www.adafruit.com/product/5477
), if you use this specific dev board then you can use the pin mappings and
the libraries and compilation have been tested.  For any other dev board
there could be small differences which could cause challenges in compiling
and using libraries.

This Biodata codebase is very small, simple, and can run on almost any
microprocessor (Arduino, ATtiny).  The ESP32 is 'way overkill' for our
simple Biodata application, large storage and memory expansion is not used
in any way.  You can potentially use any ESP32 variety but you will have to
figure out all the abbreviations, pin mappings, ensure the version has
expected functionality (for example the ESP32 S2 doesn't have Bluetooth),
and hope that the libraries (USB-MIDI, AppleMIDI) will be compatible.

You can use a dev board with any one of the S3 WROOM modules and will need
to download the right board definitions and configure for your module in
Arduino.

Sam Cusumano
Electricity for Progress <https://squareup.com/store/electricityforprogress>
***@***.***
267-230-1802

ooktown, 2022-08-22:
I will concur with Sam that using the board specified in the design will make programming and set up WAAAAY easier.

At the time we made our boards, the only Feather ESP with BLE and native USB was the [FeatherS3 by Unexpected Maker](https://esp32s3.com/feathers3.html). I had to figure out the new pin layout and re-map the differences (pretty much every pin had to change), and even at that the Adafruit MIDI over USB library was stubborn and would not compile 👎  . So in the end, I took out the MIDI over USB feature (until such time that the library becomes compatible with the FeatherS3 by Unexpected Maker).

To me, the MVP is having WiFi and BLE with a battery. Really handy! Using the Biodata interface next to the plant without MIDI or USB cabling is awesome.

electricityforprogress, 2022-08-23:
closed - send an email if you need more help!

closed 2022-08-23 by electricityforprogress