electricityforprogress/BiodataFeather #10: Biodata_v007_ESP32S3 compile error

matwan77, 2025-01-30:
Dear Sam,

I'm trying to compile the sketch using IDE 2.3.4 for ESP32-S3 Feather but during this operation receiving information as below:

"Sketch uses 1604948 bytes (111%) of program storage space. Maximum is 1441792 bytes.
Global variables use 94116 bytes (28%) of dynamic memory, leaving 233564 bytes for local variables. Maximum is 327680 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
text section exceeds available space in board
Compilation error: text section exceeds available space in board"

PLease, could you help me to understand what i do wrong?
Thank you in advance.
Mateusz

matwan77, 2025-01-30:
Thank you Sam!
It works, now need tu study everything, including how to operate menu, because clicking button only 2 diodes are changed :)
Thank you one more time for your amazing work!
With best regards.
Mateusz
thrak10@gmail.com

electricityforprogress, 2025-01-30:
Very glad to hear that it worked! Yeah you know the menu mode and button
with knob don't really work for most users, I don't understand why because
it's the production code that I've used in my devices. My suggestion is to
ignore the button and menus, make any nitty or other configuration settings
in the code base itself, and just use the knob to control the threshold.
Although I have to admit it is fun to be able to change the scaling
algorithm on the fly.

st80356, 2025-04-09:
Hello and thank you @electricityforprogress.  I'm having a lot of trouble getting the correct library versions to compile in the Arduino IDE 1.8.1 and 2.3.4.

Could you please upload a .bin of the latest version so I can flash it directly with and ESP flasher, or provide some hints on the correct library releases that will allow it to upload with a specific version of the Arduino IDE? 

Thank you again for a really cool project!

electricityforprogress, 2025-04-11:
@st80356 yes compiling is a pain, and changes to the espressif IDE breaks code (sadly).  But there is a bin in the folder  https://github.com/electricityforprogress/BiodataFeather/blob/master/Biodata_v007_ESP32S3/Biodata_Feather_ESP32_07.ino.adafruit_feather_esp32s3.bin

st80356, 2025-04-11:
Thank you for the insight and reply!

I tried that .bin file too but I'm getting no signs of life from my
boards.

Need to look at the components to troubleshoot then.

Just one last question, which version of the Arduino IDE did you use to
compile the newest .ino?

electricityforprogress, 2025-04-11:
The .bin file is very old, back from when I posed the code, I don't know the versions.  There are many many ESP32 boards, this will only work with the ESP32-S3 variety.  If you have a normal ESP32 you can try the 005 version of the code.

st80356, 2025-04-11:
Thanks so much for your time; I'll give them a try!

closed 2025-04-11 by electricityforprogress